OpenBCI/V3_Hardware_Design_Files #5: missing PL2 connector in the schematic but still on the pcb

shadowphile, 2017-06-25:
In the past I downloaded the schematic and it had two pages.  In order to make life easier I merged the two pages together.
After doing my own layout in a different schematic CAD while using the DesignSpark schematic as a reference, I came across something wrong.   I thought I had accidentally erased something when I deleted the second page.
So I erased everything, uninstalled DesignSpark, cleaned out all the residual references like in the registry, then reinstalled DesignSpark (8) and re-downloaded the Cyton files.
Now there is only one page of schematic and it does not contain PL2 even though it is on the PCB.
Now I'm afraid I might be missing other stuff that won't make it onto my own PCB if the official schematic is not complete.
thanks

biomurph, 2017-06-26:
There should be only one page in the schematic.
You may need to make a project, and then put the files into it.
I don't see PL2. What are you referring to?

shadowphile, 2017-06-26:
The pcb in SparkDesign has a PL2 in the Goto list.  Four pads on the
backside, see attached picture.  They are tied to the RFArduino as RFRX and
RFTX and RFRST but on the schematic those lines go nowhere else.
thanks

joelSensor, 2017-06-27:
Oh,
I think I made those SMT pads inside the PCB layout window. I don't think that there is a schematic part for them. You can make pads, place them, and assign nets to them in the pcb layout.

shadowphile, 2017-07-07:
ok, thanks.  as long as those RFRX and RFTX lines aren't used for anything.