OpenBCI/V3_Hardware_Design_Files #9: Daisy module broken or fixed in repo?

ak1n, 2017-08-20:
Does the current daisy module design (which I have already gotten fabbed...) include corrections re the initial design error per here: http://openbci.com/community/daisy-module-re-work/ or do I still need to get the pin corrected? Thanks!

biomurph, 2017-08-21:
This issue was fixed, and this repo should reflect that.
I have migrated from Windows to MAC a couple of years ago. Will verify when in front of a Windows machine. The pin in question is the CLKSEL, pin 52, and it should be pulled to GND so that the Daisy will accept the external CLK signal.

ak1n, 2017-08-26:
Awesome. Thank you for your response, it is appreciated.

Frederik-D-Weber, 2019-06-17:
I checked the design files with DesignSpark. They are still _not_ fixed! The issue remains.

It is more than 4.5 years later.  In the design files, the pin 52 of the ADS199 is still connected to DVDD and not to AGND as is should be. Given the 1) issues this causes, 2) mulitple requests by the community to fix the files, 3) the effort that went into decribing this fix, 4) leaving the impression here that it is fixed... it is not acceptable to have the faulty version on github. Obviously a fixed version exists, since it is used it for the new batches sold on the OpenBCI shop.

Please fix this asap, or I will fork a fixed version.

retiutut, 2022-07-25:
Thanks @tharun-OpenBCI for https://github.com/OpenBCI/V3_Hardware_Design_Files/pull/16

closed 2022-07-25 by retiutut