OpenBCI/V3_Hardware_Design_Files #11: Schematic question——BIASOUT should feedback to bias_INV?

trzp, 2020-03-09:
Hello:
I have read the ads1299 document and the cython's schematic, I found that in the ads1299 document, the "bias drive" output signal (pin63 biasout) is feedback to BIAS_INV pin , however, in the cython's schematic the bias drive signal feeds back to BIAS_IN and leave BIAS_INV float. It makes matters? In my understanding, the bias_in pin is used to connect 'BIASOUT' and a special input channel electrode with proper register configuration. So, I think whether the BIASOUT should feedback to BIAS_INV, and maybe connect the BIAS_IN to ground?

R0dri, 2020-05-14:
I recently came across with this same issue. What I gather:
- BIAS_IN is at the input multiplexer that can be mapped to any channel similar to SRB1, SRB2.
- BIAS_OUT is bridged to BIAS_IN in case you want to see what is the signal from the Bias Amplifier
- BIAS_INV is mapped to the daisy bus. Look ahh Figure 39 from the data sheet to understand why.

If you bridge BIAS_INV with BIAS_OUT you will set the integrated Bias amplifier as a buffer (signal follower). This will let you measure the precise common mode noise present. The config running on board tells me they intend to have the Bias drive amplifier (pin BIAS on the board) to be the amplified difference between the ground plane and the common mode noise collected by the PGAs array. This will effectively send an inverse signal of the common mode noise to be feed to the body. The driver is left open so this signal is as strong as the driver can amplify it in an attempt cancel the noise present. Since the noise levels are commonly very low in magnitude, saturation of the amplifier is not a concern. Are you familiar with the Driven-Right Leg circuit?

Anyhow, I guess you can still connect a jumper wire between BIAS pin and BIAS_INV on the board to accomplish the voltage follower proposed by the data sheet as you mentioned.

Here is the data sheet for reference:
(http://www.ti.com/lit/ds/symlink/ads1299.pdf)

JoshQuint, 2023-04-11:
You might also be interested in these discussions regarding the same topic:
- [TI forum](https://e2e.ti.com/support/data-converters-group/data-converters/f/data-converters-forum/424183/ads1299-bias-setting)
- [OpenBci forum](https://openbci.com/forum/index.php?p=/discussion/2917/bias-pinout)