OpenBCI/V3_Hardware_Design_Files #23: How to open .sch or .pcb file in EAGLE

shubhankar9, 2023-01-04:
Hi, I wanted to convert this design to a single or two layered PCB. I am comfortable using EAGLE and so I wanted to open the 32bit .pcb and .sch files in EAGLE. I can't seem to find a way to do that. I am able to open and view it in DesignSpark but there is not export option that makes these files compatible with EAGLE. Would love your thoughts on this.

JoshQuint, 2023-03-14:
As I had the same problem I can tell you that DesignSpark and Eagle files are not compatible. 
I could not find any tools that would allow you to import them yet.
Therefor, if you want to stick to Eagle, you will basically have to use the DesignSpark schematics as a reference to build your own schematic and later design the board in Eagle.
Alternatively, look up open-source hardware projects that might use the same ADC and Eagle.
Also keep a track of the open issues in this repository, in case you consider a different approach for your own project.

shubhankar9, 2023-03-14:
That's interesting. Good to know. Thank you!

Regards
Shubhankar