OpenBCI/V3_Hardware_Design_Files #33: Cyton BOM vs Schematic vs Design Spark Inconsistency Issue

FlywheelMist, 2024-12-10:
Hello,
I was looking through the schematics, BOM, and designspark to find that there are 3 different components and wiring for the R11, R10, D10. This includes R11 resistance value (330k, 470k or 1k), R10 resistance value (10k, 1k, 220k) and the presence of D10 entirely.

Could these please be specified or updated?

Many thanks,
M

Images:

DesignSpark:
[image]

BOM:
[image]

Schematics:
[image]

atulakella, 2025-11-16:
Hey!
Just stumbled across the same issue, I see a lot of inconsistencies in BOM and component position files, did you find a workaround or did you get help from the maintainers of this repo?

FlywheelMist, 2025-11-16:
Hey @atulakella 
Afraid I've long since abandoned this project and don't remember finding a solution... I think your best bet for this is to contact someone from openbci directly... Sorry I can't be of more help.
M